corbanmailloux/esp-mqtt-rgb-led #6: Brightness transitions can overshoot.

corbanmailloux, 2016-11-07:
If I have a light set to brightness: 255 and color: 33, 255, 19, and then I fade it to brightness: 40 with a transition of 1, it seems to overshoot and I end up with a very dim purple. 

If I send the same request again, the lights set to the correct value.

corbanmailloux, 2017-10-24:
I would think that this is caused by something with the [comparison here](https://github.com/corbanmailloux/esp-mqtt-rgb-led/blob/master/mqtt_esp8266_rgb/mqtt_esp8266_rgb.ino#L412). I haven't looked into it yet, but maybe the `lastLoop` isn't being set correctly for short transition times.

eldoner, 2017-10-31:
https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs/issues/15