corbanmailloux/esp-mqtt-rgb-led #7: White channel support

2fat2kidnap, 2016-11-08:
I just came across this project as I was looking to update my MQTT/RGBWW LEDstrip config and I did not want to reinvent the wheel. I was wondering whether it was on the roadmap for this project to add support for cool/warm white channels on RGBW(W) LED strips?   

Project looks great!

bhaap, 2016-11-17:
/this

Im looking at getting some LYT8266 bulbs and to be able to build them into my home assistant would be awesome :D

corbanmailloux, 2016-11-17:
This can certainly be added. Do either of you know an example light platform in Home Assistant that supports white channels? I'm unsure of how it works from that side.

For the ESP side, an additional JSON field could be read in `processJson(...)` and set for the lights. I don't currently have any RGBW lights, but I can structure something out. 
Is it basically just another color channel? So red: 0-255, green: 0-255, blue: 0-255, white: 0-255?

bhaap, 2016-11-17:
I think that this has it. https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/light/osramlightify.py

The Kelvin setting is a nice addon aswell. Added by running a bit of red/blue at the same time as the white led.

I added white to your current file in my branch. I have almost no experience in programming but hopefully it would work fine :)
https://github.com/bhaap/esp-mqtt-rgb-led/blob/master/mqtt_esp8266_rgbw.ino

corbanmailloux, 2016-11-30:
Hey @bhaap. Sorry for the slow response; I'm near the end of a semester at college, so other things are taking priority for now.

In the example you linked (`osramlightify.py`), I see that it supports color temperature, but I don't see how it supports a separate white channel. The reason I'm asking is I don't see any way to have Home Assistant send a white value to any platform. I know that this project is technically just the ESP side, but I'd like to keep it compatible with the `mqtt_json` platform I built for Home Assistant.

bhaap, 2016-11-30:
How about this? https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/light/limitlessled.py

Im no expert but there must be some way to get this to work :D

Just got my hands on a LYT8266 that would fit perfectly with this project. ESP8266 + RGBW inside of a lampsocket. its perfect! :D

bhaap, 2016-12-01:
Hmm im thinking that the call of color temperature is actually the White channel. 
Looking at this https://home-assistant.io/components/light.mqtt/
There is a Color temp command and it could be whats the controlling the white channel.

Tell me if im way off :P

bhaap, 2016-12-01:
def turn_on(self, transition_time, pipeline, **kwargs):
        """Turn on (or adjust property of) a group."""
        from limitlessled.presets import COLORLOOP
        # Check arguments.
        if ATTR_BRIGHTNESS in kwargs:
            self._brightness = kwargs[ATTR_BRIGHTNESS]
        if ATTR_RGB_COLOR in kwargs:
            self._color = kwargs[ATTR_RGB_COLOR]
        # White is a special case.
        if min(self._color) > 256 - RGB_BOUNDARY:
            pipeline.white()
            self._color = WHITE
        # Set up transition.
        pipeline.transition(
            transition_time,
            brightness=_from_hass_brightness(self._brightness),
            color=_from_hass_color(self._color)

bhaap, 2016-12-14:
Hello, sorry to disturb you in the end of your college semester but have you found a way to utilize the white channel in limitlessled? I myself are completely lost here :D

bhaap, 2017-01-24:
Hello, im wondering if you have made any headway with this? Currently im using a mqtt program that i wrote myself that has RGB + W + "Party-mode" and with that im splitting it up into 3 different lights to control them. And have hardcoded into the code so that only one can be on at a time.

I would be nice to have these all in one :)

mattues, 2017-06-20:
Any updates for white channel yet?

corbanmailloux, 2017-07-31:
Hey everyone here. I'm sorry I haven't been following up here. I've reviewed the updates for the `mqtt_json` light platform on Home Assistant, and I've added a new version of the code that supports the white value. 

Here's the proof:
[image]

See the new folder/file called [mqtt_esp8266_rgbw](https://github.com/corbanmailloux/esp-mqtt-rgb-led/tree/master/mqtt_esp8266_rgbw).

Thanks for your patience.

closed 2017-07-31 by corbanmailloux