corbanmailloux/esp-mqtt-rgb-led #10: Way too low brightness

ToxicF3AR, 2017-02-22:
I created this project today with Normal 12 DC RGB Led and IRF840N Channel Mosfet but this brightness is way too low any help ?

corbanmailloux, 2017-02-22:
So if you bypass the MOSFET and power the LEDs directly, they are much brighter?

I'm not sure what would cause that. I can't look into the datasheets at the moment but are you sure that the MOSFET is turning on at 3.3v? What if you attach the MOSFET directly to +3.3v instead of running through the ESP?

ToxicF3AR, 2017-02-23:
I did it just as it is described in the picture

corbanmailloux, 2017-02-23:
Could you test the things I suggested in my reply?

ToxicF3AR, 2017-02-23:
I did the same thing as it is on your pciture and in your picture it is powered directly from 3.3v rail not from esp so yes I powered it directly from 3.3v rail not from esp

corbanmailloux, 2017-02-23:
Please test the MOSFETs. Here's what I'd suggest:
Based on my picture, take the red, green, and blue wires that connect the MOSFETs to the ESP. Connect them from the MOSFETs directly to the 3.3v rail. That should force the MOSFETs to all turn on, setting the LEDs to maximum brightness.

ToxicF3AR, 2017-02-24:
Sorry but I cant understand what are you trying to say can you edit the picture and show me what you are trying to say?

corbanmailloux, 2017-02-24:
I don't have the time right now to make a special wiring diagram just for this. Simply connect the gate leg of each MOSFET to the 3.3v rail, turning on the MOSFETs.

corbanmailloux, 2017-02-26:
Connecting the MOSFETs directly to the 3.3v rail was just to test that the MOSFETs were functioning. By wiring it that way, we bypassed the ESP all together. 

Now that we know the MOSFETs are functional and the brightness is fine, return the wiring to normal. 
Are you able to directly send MQTT messages to the device using an MQTT client?

ToxicF3AR, 2017-02-27:
I didnt tried to send MQTT Messages using an MQTT Client but I used Homeassistant and its worked fine but due to normal wiring the brightness is low.

corbanmailloux, 2017-03-17:
@ToxicF3AR Did you come up with a solution for this issue? I haven't been able to reproduce the issues with my setup and I haven't heard from others with the same issue.

ToxicF3AR, 2017-03-17:
The solution of this is only to replace the mosfets with the one you are using.

closed 2017-04-17 by corbanmailloux

osmantuna, 2018-02-05:
I am using IRFZ44N as mosfet and i have exactly same issue. Full brightness when bypassed but like 40% when normal. Switching mosfets does work?

osmantuna, 2018-03-19:
I switched mosfets with LB8721 and it works perfect. Issue with this, mosfet Vgs value must be in range.

corbanmailloux, 2018-03-20:
@osmantuna Thanks for following up. I wasn't able to reproduce this issue with the hardware I had on hand, so I was assuming that it was just the MOSFETs not matching the specs needed for being driven by the ESP8266.