corbanmailloux/esp-mqtt-rgb-led #16: Add gamma correction

makzdot, 2017-04-16:
Hi,

Thank you for writing and sharing this code. I do have one enhancement if possible. I'm new to the ESP&Arduino in general so am not able to contribute myself (yet). What i noticed that when setting the LED to yellow (i.e. 	#FFFF00). The light produces a more green then yellow light. And in general the true colors are somewhat "off".

So I think a gamma correction could/should be added. I found something in the likes of this:
https://learn.adafruit.com/led-tricks-gamma-correction/the-quick-fix

If this is what is needed it should fairly easy to implement. What do you think?

Thanks & kind regards,
Max

corbanmailloux, 2017-04-17:
Hmm... This is really interesting, @makzdot. I've struggled with this issue in the past, and I hadn't seen such a simple fix.

I'll take a look at adding this solution to the project. I'll probably leave it as a configuration option (or an entirely different file) because some lights might not exhibit the same issue.

I'm working on some other projects right now, but I will leave this issue open to remind me soon. 

If you, or anyone, would like to open a PR for this change, I welcome it.

makzdot, 2017-04-17:
Hi @corbanmailloux,
Thanks for your response. In the meantime I've been experementing and made a tinsy bit of progress. With this solution the colors are a little bit more accurate. Still not perfect though. But the oranges and yellows match up way better to actual RGB codes than before. And that is a win.
I'll make a pull request soon.