corbanmailloux/esp-mqtt-rgb-led #19: Taking #RRGGBB hex string and using it to create the JSON block with decimal values

linker3000, 2017-05-28:
One for the notes/wiki??

I was using a color wheel icon/function in an Android-based MQTT app to test my RGB LED build with _esp-mqtt-rgb-led_ and Node-red. I created this node function to turn the received hex MQTT payload (#RRGGBB) from the app into an input string for this code on an ESP8266. Might be useful for others...

`// Convert #RRGGBB hex payload into JSON formatted string for
// corbanmailloux/esp-mqtt-rgb-led code
// https://github.com/corbanmailloux/esp-mqtt-rgb-led

if ((msg.payload.toString().length != 7) &&
    (msg.payload.toString().substring(0.1) != "#"))
  {return null;}

msg.payload = '{"state": "ON","brightness": 255,"color": {' + 
  '"r": ' + parseInt(msg.payload.toString().substring(1,3),16) + ',' + 
  '"g": ' + parseInt(msg.payload.toString().substring(3,5),16) + ',' +
  '"b": ' + parseInt(msg.payload.toString().substring(5),16) + 
  '},"transition": 2' +
  '}';

return msg;`

corbanmailloux, 2017-05-29:
@linker3000 Thank you for this addition. I've added a wiki page [here](https://github.com/corbanmailloux/esp-mqtt-rgb-led/wiki/Helpful-Notes) and credited you.

closed 2017-05-29 by corbanmailloux

linker3000, 2017-05-30:
Thanks! In further testing I modified the code slightly:

```
// Convert #RRGGBB hex payload into JSON formatted string for
// corbanmailloux/esp-mqtt-rgb-led code
// https://github.com/corbanmailloux/esp-mqtt-rgb-led

if ((msg.payload.toString().length != 7) ||
    (msg.payload.toString().substring(0,1) != "#"))
  {return null;}

msg.payload = '{"state": "ON","brightness": 255,"color": {' + 
  '"r": ' + parseInt(msg.payload.toString().substring(1,3),16) + ',' + 
  '"g": ' + parseInt(msg.payload.toString().substring(3,5),16) + ',' +
  '"b": ' + parseInt(msg.payload.toString().substring(5),16) + 
  '}' +
  '}';

return msg;
```