corbanmailloux/esp-mqtt-rgb-led #20: Add the ability to updates ESP8266 via OTA / OTA Uploading

ctripodi, 2017-06-17:
A features that could be great to have in this excellent project, is the  ability to updates ESP8266/NODEMCU via OTA / OTA Uploading
http://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html
Thanks.

corbanmailloux, 2017-12-07:
I did the work for this on the [add-ota-upload](https://github.com/corbanmailloux/esp-mqtt-rgb-led/compare/add-ota-upload) branch, but I could never get the OTA upload reliably working. When the work on the unified version of the code is complete (see #31), I'll take a look back at this and see if I can finally get it working.

Philje123, 2017-12-14:
I've successfully added OTA code to your code and it works perfectly.

I use it on a Magic Home RGB wifi controller. Other firmwares were too complicated for my needs so I've used this and changed the data pins accordingly and now added OTA.

Added this to config:

`/**************************** FOR OTA **************************************************/
#define SENSORNAME "kitchenislandleds" //change this to whatever you want to call your device
#define OTApassword "supersecretpassword" //the password you will need to enter to upload remotely via the ArduinoIDE
int OTAport = 8266;`

Then just added this between setup_wifi and void setup_wifi:

` //OTA SETUP
  ArduinoOTA.setPort(OTAport);
  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname(SENSORNAME);

  // No authentication by default
  ArduinoOTA.setPassword((const char *)OTApassword);

  ArduinoOTA.onStart([]() {
    Serial.println("Starting");
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}`

corbanmailloux, 2017-12-14:
@Philje123 Thanks for sharing. I've been planning to come back to the OTA work soon, and that example might help. I'm not sure if it's much different that what I had before, but I'll review and get it working. 

(Note: I edited your comment just to add code formatting. When adding a multiline code block, surround the code with ` ``` ` on the lines above and below the code.)