corbanmailloux/esp-mqtt-rgb-led #24: Feican LED Bulb flashing

pickeld, 2017-07-29:
hey, i was able to flash your code to the lamp and control it using MQTT, thank you for your hard work!
my Feican lights are RGBW which means i can use the warm white lights too.
in your code then is no such option, do you think its possible to add the ability to control the white lights too?

thank you and have a great week

Landrash, 2017-07-30:
@pickeld se #7 . Duplicate request.

corbanmailloux, 2017-07-31:
Hey @pickeld. I'm going to close this as a duplicate of #7. Please follow the conversation there.

closed 2017-07-31 by corbanmailloux

corbanmailloux, 2017-08-02:
Hi @pickeld. As you'll see in #7, I've added white channel support for this project. If you get a chance to, please test out the new version with your system and let me know how it's working.

pickeld, 2017-08-11:
you are awesome! working perfectly! thank you :)

corbanmailloux, 2017-08-12:
Good to hear!