corbanmailloux/esp-mqtt-rgb-led #29: Merge scripts into one

r41d, 2017-08-31:
Although this would introduce some (ugly) #ifdef and #ifndef I think it makes sense because 90% of the code is shared and it is very tedious to add features in all three files.

corbanmailloux, 2017-08-31:
I've actually been wanting to do this. It was initially just one script, but then I made a new one for one specific light and then we ended up where we are now. 
If you'd like to take a first pass at combining them, I welcome it. I've been a little too busy on some other projects to do much with this one.

r41d, 2017-08-31:
If I were to do this, I would leave as much of the code doing RGBW logic and only use the #ifdef stuff where it's relevant for communication with the outside world (i.e. when receiving messages, publishing them, or accessing the IO pins of the ESP).
Does that sound reasonable?

corbanmailloux, 2017-08-31:
Yeah, I think that makes sense. I'm worried that the code will become pretty messy, but maybe I'm overestimating that.

corbanmailloux, 2017-12-11:
Closed with #31.
One last step is tracked in #36.

closed 2017-12-11 by corbanmailloux