corbanmailloux/esp-mqtt-rgb-led #30: Enhancement request

Ryanmaturi, 2017-09-04:
Add Gpio inputs to physically turn on and off the light, incase one does not have access to phone. this could be one momentary switch which would increase brightness or decrease if max bright is already met. or if pressed briefly could turn off light. not sure how you would handle RGB

corbanmailloux, 2017-12-07:
Thanks for the request. The one limitation is that I have been using the 8-pin ESP8266 boards (like in the read me) for these lights, so I'm pretty much out of GPIO pins.
I've since started using Wemos D1 Minis as my default board for this kind of project, and I can definitely see cases where a button would be helpful.

When I have time available, this is a feature that I'd like to add. Alternatively, if anyone would like to make the change and open a PR, I welcome it.

depuits, 2018-09-24:
I've added button compatibility in my [fork](https://github.com/depuits/esp-mqtt-rgb-led). You can use it with other esp modules such as the esp-12 of when you're only using a single color led strip. @corbanmailloux if you like the implementation I would like to make a PR.