corbanmailloux/esp-mqtt-rgb-led #32: mqtt light (2nd entry) not showing on home assistant

AAli77, 2017-09-18:
light:
  - platform: mqtt_json
    name: mqtt_json_light_1
    state_topic: "home/rgb1"
    command_topic: "home/rgb1/set"
    brightness: true
    rgb: true
    effect: true
    effect_list: [colorfade_slow, colorfade_fast, flash]
    optimistic: false
    qos: 0

    name: mqtt_json_light_2
    state_topic: "home/rgb2"
    command_topic: "home/rgb2/set"
    brightness: true
    rgb: true
    effect: true
    effect_list: [colorfade_slow, colorfade_fast, flash]
    optimistic: false
    qos: 0

Can you/anyone please assistant me what I am doing wrong.

I had compile the code no error, home assistant load perfectly but LED Switch is showing only for mqtt_json_light_2.

thanks

AA

corbanmailloux, 2017-09-19:
@AAli77 For future reference, when you post your code on Github you can wrap it in three backtick (<kbd>\`</kbd>) characters to make Github format it as code and keep the original formatting. You can also use the code button in the comment editor (<kbd>< ></kbd>). For example:
````
```
# CODE
```
````

To your question:
[Here's a sample of multiple `mqtt_json` lights from my own configuration](https://github.com/corbanmailloux/home-assistant-configuration/blob/e6b5286571c5865b919d669a0bdc6e6b8f74ad3b/light/mqtt_json.yaml#L3-L21).

If I'm reading your configuration correctly, it looks like you are only listing the `platform` once and you have both lights in the same list item. Note how in my configuration the `- platform: mqtt_json` line is repeated at the beginning of each light. That lets Home Assistant know you are creating a second light.

closed 2017-09-19 by corbanmailloux

AAli77, 2017-09-19:
Thank you Sir, I will try your code. And also thanks a lot for your quick response