corbanmailloux/esp-mqtt-rgb-led #35: Plans for development?

moskovskiy82, 2017-12-06:
Seems this is the best RGB code for homeassistant at the moment. So wonder if there are any plans for further development?
Separate control for White light in RGB? More effects?

depuits, 2017-12-07:
I was thinking to create an interface for the effects and moving them to separate header files. This would make it easier to add new effects without the need of adding new global variables for each of them.

moskovskiy82, 2017-12-07:
That would be great. Christams is coming and it would be great to add some effects for that

corbanmailloux, 2017-12-10:
@moskovskiy82 Thanks for the interest in this project. After taking some time off, I'm back to working on my home automation projects like this one. 

The project does support having a separate white channel (from [issue #7](https://github.com/corbanmailloux/esp-mqtt-rgb-led/issues/7#issuecomment-318954698)). I am updating to README tonight to document the latest improvement, #31, which combines all of the individual options into one file. This will make it a lot easier to add features going forward.

I would love to see more effects supported. I like @depuits' suggestion of breaking out the support for different effects. If you (@depuits) have any interest/time to work on that, I always appreciate the help. 

I also plan to look into FastLED to see if any of the calculations or features it provides could be helpful here. I don't particularly like to reinvent work that others have already perfected.

I will try to keep the [issues list](https://github.com/corbanmailloux/esp-mqtt-rgb-led/issues) up to date as the correct place to track planned work and work-in-progress.

closed 2017-12-10 by corbanmailloux