corbanmailloux/esp-mqtt-rgb-led #37: White value RGB

moskovskiy82, 2017-12-11:
As the previous one was closed o will reopen.
Currently for RGBW there is a white value in HA. This controlls the the white diodes. But it only works together with RGB. So there is no way currently to switch on white diodes without switching on RGB. Exposing W value as a separate light will do the trick

corbanmailloux, 2017-12-11:
(Note: This discussion is continued from #35)

I think I understand the issue you're having (please check that the issue title is correct).
I'll experiment with this. I don't really like the idea of exposing two lights in Home Assistant for this purpose, when Home Assistant already supports the white channel. Let me play with the interaction between this code and Home Assistant to see what the options are.

moskovskiy82, 2017-12-11:
Yes! Thank you. The title reflects it. I can provide a link to github project which i have used previously. There it exposes 2 lights. Not the best approach either but might it help?

corbanmailloux, 2017-12-11:
If you have a reference you could share, that might be helpful.

Thanks.

moskovskiy82, 2017-12-11:
https://github.com/open-homeautomation/h801 this is the project i have used for H801 which is a chinese ESP8266 based rgbww module

moskovskiy82, 2018-01-10:
Any hope on progress?

corbanmailloux, 2018-01-15:
I'm still not sure that this is the right approach. If this is added, I'll include it as a configuration option. 
I'm not actively working on this proposal at the moment, but I'll continue to track it here as an issue.
#43 introduces some major changes and rework, but after that is merged I'll look into the effect of adding a configuration option for this. I'd also welcome a PR adding this.

Thanks for you patience.