corbanmailloux/esp-mqtt-rgb-led #44: adding support for WW/CW led strips

whiney1, 2018-02-10:
Hi corbanmailloux, thanks for this project, I've used it for a couple rgb strips and it's great!

I'm now hoping to use it with a new led strip that has Warm White/Cool White led chips run on separate wiring loops like regular rgb chips. Doing some research, it looks like:

- your code has one white channel already
- the MQTT JSON component already has colour temperature support built in

Is it possible to add support for a dual white mode to this firmware, so that I can then add a 'single' colour light to homeassistant that is brightness and colour temp variable? 

Cheers!

corbanmailloux, 2018-02-16:
Thanks for the suggestion. This is something I've thought about before, and it would be valuable to add.

I'll keep it tracked in this issue and update when I (or anyone) get a chance to start working on it.

whiney1, 2018-02-21:
Cool!