corbanmailloux/esp-mqtt-rgb-led #54: Can this be used with cloud mqtt

Ameengray, 2019-02-08:
HI is this limited to Home Assistant, can we use this with cloud mqtt ?  I see only one topic when the device connects to the server, what messages do we have send to change the state of the device ?

depuits, 2019-02-11:
In the code you can find these sample payloads to adjust the state of the device.

```
SAMPLE PAYLOAD (BRIGHTNESS):
{
  "brightness": 120,
  "flash": 2,
  "transition": 5,
  "state": "ON"
}
SAMPLE PAYLOAD (RGBW):
{
  "brightness": 120,
  "color": {
    "r": 255,
    "g": 100,
    "b": 100
  },
  "white_value": 255,
  "flash": 2,
  "transition": 5,
  "state": "ON",
  "effect": "colorfade_fast"
}
```

This example can be found [here](https://github.com/corbanmailloux/esp-mqtt-rgb-led/blob/master/mqtt_esp8266_light/mqtt_esp8266_light.ino#L138) and the actual parsing of the json currently happens [here](https://github.com/corbanmailloux/esp-mqtt-rgb-led/blob/master/mqtt_esp8266_light/mqtt_esp8266_light.ino#L198).

corbanmailloux, 2019-02-15:
Thanks for responding, @depuits. The other place I would suggest taking a look at is the [Home Assistant docs for the JSON light](https://www.home-assistant.io/components/light.mqtt/#json-schema). Even if you aren't using Home Assistant, it does document the payload format.

And to answer the overall question: Yes, I built this initially with Home Assistant in mind, but it should be perfectly usable with anything else that can speak MQTT.

closed 2019-02-15 by corbanmailloux