corbanmailloux/esp-mqtt-rgb-led #59: Ability to use only white led

fusionstream, 2020-05-08:
Can you add the ability to control the white led while turning off all RGB leds. I could've sworn I opened an issue before where I either provided a full solution or someone else did but I can't find it. I think the colour wheel may have had a black option before but I can't find that either.

The simplest solution I can rethink off right off the top of my head is to change the code in the example sketch on line 183 from:
`realWhite = map(white, 0, 255, 0, brightness);`
to:
`realWhite = map(white, 0, 255, 0, 255);`

This maps the white value without regard to the brightness value which allows one to individually control RGB and WHITE.

In lieu of a better solution, please consider adding this modification to the example sketch.

corbanmailloux, 2020-05-08:
This sounds like a duplicate of #37. Is that correct?
Honestly I haven't worked on this project in a while, but I thought I resolved that at some point.

Now that we're in COVID isolation mode, I'll break out an ESP again and do some testing.

fusionstream, 2020-05-09:
> This sounds like a duplicate of #37. Is that correct?
Seems about right but isn't something I think I've seen before

> Honestly I haven't worked on this project in a while, but I thought I resolved that at some point.
See, even you thought this was settled! lol

The solution provided above does work btw. I'm using it now. It's still not completely separate because effects will still kill the white led.