corbanmailloux/esp-mqtt-rgb-led #60: JSON message length

jaffons, 2020-05-11:
Hello. I have implemented this to ESP32 with simple cp-paste from example sketches. Basically I changed:
#include <ESP8266WiFi.h> to #include <WiFi.h> 
analog.Writes to corresponding ledc library.

JSON library is intact as it should be.

However for some reason callback() function isn't called if JSON message is too long. 6 lines/objects is fine, 7 or more isn't. 

I did found that message goes nicely to thru MQTT broker and that it doesn't matter which objects are in the payload. Other than that I'm out of ideas.

corbanmailloux, 2020-05-12:
I haven't seen this behavior, but I also haven't tested this code on an ESP32. 

Have you tried increasing the JSON buffer size by increasing the value here?: https://github.com/corbanmailloux/esp-mqtt-rgb-led/blob/master/mqtt_esp8266_light/mqtt_esp8266_light.ino#L27

jaffons, 2020-05-12:
Yeah I tried to increase that one also.

Even the JSON buffer is too short this shouldn't have effect on callback(), I think. 

MQTT default message size is 128 bytes which should be plenty.

jaffons, 2020-05-12:
So indeed I made local change to PubSubClient.h line 29 
#define MQTT_MAX_PACKET_SIZE 512

...And now it works as intented. This problems seems to be known for long.

I wonder how you've not seen this before because this is not ESPxxx relevant issue and mqtt packet is almost always >126 bytes long..